sbcshop/MotorShield #4: I am trying to make a robot, but i can't quite get the code to work.

WarriorOfScrubs, 2017-12-07:


sbcshop, 2017-12-08:
Hi,

Please let us know which code is not working.

GAJENDER SINGH
DIRECTOR
<02035140914>02035140914
<#UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_>
gajender.singh@sb-components.co.uk
https://shop.sb-components.co.uk
<https://twitter.com/sb_ltd> <https://www.facebook.com/sbcomponentsltd>
<https://www.youtube.com/channel/UC-HPa0xV9LtjqT79aajhGfQ>
<https://plus.google.com/115397217718246801767>
<https://shop.sb-components.co.uk>
Unit B, 58B Kingsbridge Crescent,
Southall,  UB1 2DL  United Kingdom
Vat: GB154489384

This electronic transmission is strictly confidential and intended solely
for the addressee. It may contain information which is covered by legal,
professional or other privilege. If you are not the intended addressee, you
must not disclose, copy or take any action in reliance of this
transmission. If you have received this transmission in error, please
delete it and notify us as soon as possible. The Internet cannot guarantee
the integrity of this message, SB Components Ltd will not therefore be
liable for the message if modified.

WarriorOfScrubs, 2017-12-08:
On the motor test program, the raspberry pi responds by saying there is no PiMotor module.