sbcshop/MotorShield #7: Motors 1 and 2 not running

AdamSchilperoort, 2018-03-23:
I've followed tutorial after tutorial to run a robot using a motor-shield and these dependencies.  When I run any of these test files, only Motor 3 and 4 trigger. Any ideas?

ankurSB, 2018-03-23:
Have you tried Test_Motor.py ? Or if you have multimeter, measure voltage level at Motor Connector M1 and M2

acondra2, 2018-03-31:
I am having the same issue. My input voltage is 19v, motor 1 port output voltage is 2.25v. It's like it is still trying to draw its power from the pi instead of the attached power supply.

sbcshop, 2018-05-25:
Hi acondra,

This might be due to some hardware problem. Or may be Motor Driver IC is not working properly.

closed 2018-05-25 by sbcshop

ProudRoach, 2018-10-30:
My motor1 will run my 6v motor in reverse but not forward. I'm using the test_motor.py
Is there a chance that something could be broken so it won't run my motor forward? Is the code not right for using a 12v battery with a 6v motor? I also have something between the shield and my motor that makes it run from 12 to 6V

sbcshop, 2018-10-30:
Hi ProudRoach,

Please test your MotorShield GUI with the code. You may follow https://github.com/sbcshop/MotorShield/blob/master/MotorUI.py for the GUI code and https://github.com/sbcshop/MotorShield/blob/master/Motor_Shield_GUI_DOC.docx for the reference documentation.

In case this doesn't resolve your issue, please get in touch at our support: support@sb-components.co.uk