sbcshop/MotorShield #10: Using IR Sensors

ehansi, 2018-08-28:
Hello,

I bought an analog IR distance sensor (https://www.robotshop.com/en/sharp-gp2y0a21yk0f-ir-range-sensor.html) and wanted to test it on the motor-shield. I've connected VCC to +, GND to -, and V0 to OP on the board. Running the provided test program there seems to be no reaction, IR is never triggered.
Tested on both IR connectors. I'm pretty new to all of this so I guess I'm missing something very basic. Thanks for any help, people!

sgarret3, 2018-12-17:
My problems were because something is backwards here for the IR sensors only. When I use this example code:
#-------------------------------#
from PiMotor import Sensor
ir1 = Sensor ("IR1",0)
ir1.trigger()
if ir1.Triggered:
   print("Obstruction Detected")

#-------------------------------#

This code would only enter the IF statement when there was NOT an obstruction, which was causing me misery when I kept putting something in front of the sensor, and it would never enter the IF statement. I had to rewrite the code to follow this form:

#-------------------------------#
from PiMotor import Sensor
ir1 = Sensor ("IR1",0)
ir1.trigger()
if ir1.Triggered:
   print("Nothing Detected")
   # keep going on like normal#
else:
  print("Obstruction Detected")
  #code for when IR sees an obstruction#
#-------------------------------#

PiMotor.py has an IR pin voltage backwards (checking for pin voltage up when they mean down or vice versa) but I don't know how to fix it. This also applies to IR2 input.

If you just want a piece of code as an example. Here is a very unfinished, untested, (and poorly written) chunk of something I am preparing for some  kids to demonstrate a little tank navigating a simple maze.

[Tank.txt](https://github.com/sbcshop/MotorShield/files/2688076/Tank.txt)

mc03225, 2019-04-14:
Hi,
Take a backup of PiMotor.py, then change line 233 
if input_state != True:  (it was set to equals ==)
Now try
while True:
    if ir1.iRcheck():
        print("Detected")
    else:
       print("Not Detected")

I hope this helps?