sbcshop/MotorShield #12: Nodejs package request

mlnima, 2019-01-13:
Hi,

i wanna write with nodejs but i dont know which motor is on which pin in gpio, can tell me pin numbers ?

Thanks

GaltresIT, 2019-02-03:
motorpins = {"MOTOR4":{"config":{1:{"e":32,"f":24,"r":26},2:{"e":32,"f":26,"r":24}},"arrow":1},
                 "MOTOR3":{"config":{1:{"e":19,"f":21,"r":23},2:{"e":19,"f":23,"r":21}}, "arrow":2},
                 "MOTOR2":{"config":{1:{"e":22,"f":16,"r":18},2:{"e":22,"f":18,"r":16}}, "arrow":3},
                 "MOTOR1":{"config":{1:{"e":11,"f":15,"r":13},2:{"e":11,"f":13,"r":15}},"arrow":4}}

GaltresIT, 2019-02-09:
I have built you an example using nodered on how you can control your motors. see - [https://flows.nodered.org/flow/1a7da31268990e01902131db3bb28445](url)

mlnima, 2019-02-09:
Hi Jonathan ,

thank you so much for caring about my request :) , i am web developer and
using nodeJs as backend , i wanna build big robot by using many Rasberypi
together . is there any way you make some package for node js not the red
node ?

Thanks in advance  and Best Regards

GaltresIT, 2019-02-10:
You can use ... https://github.com/fivdi/pigpio. Replace the example with
the pins in my comment in your Github issue and build from there. Let me
know if you have any questions.

Jonathan

GaltresIT, 2019-02-10:
Please close this issue

closed 2019-03-14 by sbcshop